JochiSt/OpenFlowMeter #32: Looking for referenced documentation

pietervans, 2024-04-10:
I am an engineer working at CERN in the LHC gas group (EP-DT-FS).
One of my responsibilities is related to the PT100 flow meters and I am doing some research on this topic.

I am contacting you because I have trouble finding the document:
`[1] M. van der Klis. PT100 Channel Flowmeter for the LHC Experiments, Technical Student Report Feb. 2001`

I have also seen the document cited with an extra space in "flow meter":
`Report of M. van der Klis "PT100 channel flow meter for the LHC experiments", report February 2001-02-07`

Since you cite the document in the [References section](https://github.com/JochiSt/OpenFlowMeter?tab=readme-ov-file#references) of the README file: could you share this document with me?
You can share it here or use my email address: pieter.vanslambrouck@cern.ch

I would really appreciate it if you could share the document!

JochiSt, 2024-04-10:
I've sent you the documentation via Email since I do not know, whether I'm allowed to upload it here.

You are absolutely right, the title is "PT100 channel flow meter for the LHC experiments". I'll fix it in the README.

JochiSt, 2024-04-10:
I added a link to the documentation (stored at CERNbox)

closed 2024-04-10 by JochiSt